ros-noetic-arch/ros-noetic-desktop-full #1: Issue Tracker for Noetic Release on Arch Linux

acxz, 2020-05-02:
@bionade24 @jwhendy @stertingen
Hey guys for right now let's use this issue to track the discussion on the release of the noetic packages. Here is the project dashboard: https://github.com/orgs/ros-noetic-arch/projects/1

bionade24, 2020-05-02:
Isn't there some new GH chat feature?

acxz, 2020-05-02:
There is Gitter if that is what you are talking about: https://gitter.im/

bionade24, 2020-05-02:
No, I don't mean gitter. That's something obselete from Gitlab. There is GH chat as beta.

acxz, 2020-05-10:
@bionade24 would you be okay if I start adding packages to the AUR and making you the maintainer of them?

acxz, 2020-05-20:
@bionade24 sorry to ping u again, but would like to know your stance on this so that we can go ahead and start uploading packages to the AUR.

acxz, 2020-05-23:
I'll go ahead and start syncing the packages under my name for now then. We can transfer ownership later if we need to.

acxz, 2020-05-24:
Closing this issue for now, since an initial sync has been made. Many packages still need to have their melodic references changed to noetic as of right now. We'll slowly tackle them.

closed 2020-05-24 by acxz