ros-noetic-arch/ros-noetic-desktop-full #4: rosdep keys

stertingen, 2020-05-19:
Currently `rosdep` is not really usable on arch linux because lots of keys are simply missing. Maybe we could try to fill the gaps in https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml and https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml.

Having such mappings available could come in handy when automatically generating and updating `PKGBUILD`s from `package.xml` files. Right now we barely verify our dependencies as long as the package builds.

(Background: I'm currently exploring ways to generate `PKGBUILD`s for *all* packages listed here: https://github.com/ros/rosdistro/blob/master/noetic/distribution.yaml. An advantage would be that we could have an exhaustive and up-to-date set of `PKGBUILD`s.)

acxz, 2020-05-20:
Yeah this is very appealing.
However, see: https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md#arch-linux
and the following issue thread culminating at https://github.com/ros-infrastructure/rosdep/pull/559#issuecomment-631443014

In order for us to use rosdep on Arch properly we need to ensure that the packages can be installed through `pacman`, which means they need to be in the official repos or in a 3rd party repo such as arch4edu. 

For us to add the relevant package entries to the rosdep keys we need to ensure that we have the associated binaries packages. And for us to have the binary packages we need them on the AUR so that we can pull them in and package them on the arch4edu.

And then we go to rsodep and try to convince the rosdep maintainers that since these packages are on a 3rd party repo it is fine to add them to the rosdep entries (which I am not 100% sure they will do, since 3rd party repos are not supported by Arch Linux officially which was the main reason why AUR helper support was not added to rosdep)

TLDR it will be a long process, but since providing packages through arch4edu is on our list anyway, we might as well work towards this goal as well.

acxz, 2020-07-02:
Also as for your background take a look at https://github.com/ros-arch specifically https://github.com/ros-arch/ros-package-tools There is some stuff there that you can build off of.

stertingen, 2020-07-05:
With the help of [Repology](https://repology.org/) and some trivial guessing I manage to get pretty good rosdep keys using a small python script: https://github.com/stertingen/rosdep-with-aur 

Although the results are not perfect, they might provide a usable starting point to provide own rosdep keys.

If this looks good to you, this repo can be transferred into [ros-arch](https://github.com/ros-arch). To make the rosdep keys available to users, we could provide an AUR package with a `/etc/ros/rosdep/sources.list.d/10-arch.list` file like this: https://github.com/stertingen/rosdep-with-aur/blob/master/10-arch.list, rosdep will then pull our custom rosdep keys automatically.

acxz, 2020-07-05:
Yeah let's put it in ros-arch, let me give you access and we can transfer this issue over there.

stertingen, 2020-07-05:
> Yeah let's put it in ros-arch, let me give you access and we can transfer this issue over there.

Looks like we cannot move issues between organizations.

acxz, 2020-07-06:
ah yes do you mind just closing this one and creating a new issue over there then?

closed 2020-07-06 by stertingen