ros-noetic-arch/ros-noetic-desktop-full #11: [Package Request] Franka Emika packages

liborw, 2020-05-29:
Please add Franka Emika packages, I will add the PKGBUILDs, thanks.

https://github.com/frankaemika/libfranka:
 - [ ] ros-noetic-libfranka

https://github.com/frankaemika/franka_ros: 
 - [ ] ros-noetic-franka-ros (split package)

https://github.com/ros-planning/panda_moveit_config:
 - [ ] ros-noetic-panda-moveit-config

acxz, 2020-05-29:
@liborw thx for the package request for `ros-noetic-franka-ros` you will have to list the packages need for it as well, such as dependencies that are not packaged. Makes it easier for me to track and create the repos.

liborw, 2020-06-01:
@acxz do you mean dependencies that are not already in ros-noetic-arch or pacman repositories, these are only the `libfranka` and `panda-moveit-config`.

acxz, 2020-06-01:
I meant the individual packages of the metapackage, but after seeing you PR. It is clear.

acxz, 2020-06-02:
packages created and synced!

closed 2020-06-02 by acxz