ros-noetic-arch/ros-noetic-desktop-full #14: Failure while building

noahares, 2020-08-12:
I tried installing ros-noetic-desktop-full on my arch system with yay and everything went smooth until the build began. I used the -git versions as stated on the AUR page. 

This is the output before the build terminates:

```Scanning dependencies of target rosconsole
[ 83%] Building CXX object CMakeFiles/rosconsole.dir/src/rosconsole/rosconsole.cpp.o
make[2]: *** No rule to make target '/usr/lib/libconsole_bridge.so.0.4', needed by 'devel/lib/librosconsole.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:690: CMakeFiles/rosconsole.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
error making: %!s(func() string=0x55851b375bb0)
```

Is there an easy fix to this?

acxz, 2020-08-12:
If you are using arch4edu, please disable it from your `/etc/pacman.conf` and then rebuild `ros-noetic-desktop-full` as follows: `yay -Syu ros-noetic-desktop-full --rebuildtree`
That should do the trick.

noahares, 2020-08-12:
I did use arch4edu. Will try it right away. Thanks for the fast feedback.

closed 2020-08-12 by noahares

acxz, 2020-08-12:
@noahares I assume you got it to work?

noahares, 2020-08-12:
Yes, took a long time building, but everything is working now.

acxz, 2020-08-12:
Glad to hear! We are still working on getting the binary packages through arch4edu up to snuff. Sorry about the trouble you faced.

noahares, 2020-08-12:
No problem. I am really happy it finally works on Arch. Now I can get rid of my Ubuntu partition.