ros-noetic-arch/ros-noetic-desktop-full #17: Missing packages for noetic distribution

dyamon, 2021-01-10:
I'm not sure this is the correct way of notifying this but some packages for the noetic distribution  are missing from the AUR.

I'm following the [Getting Started](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) section of the MoveIt introduction. The `rosdep` command gives me a few packages that I need to manually install from the AUR but a couple of them are not present, namely:
+ `ros-noetic-moveit-resources-panda-moveit-config`
+ `ros-noetic-rviz-visual-tools`

They seem to be available for melodic (and the latter for kinetic as well) but not for noetic.

acxz, 2021-01-10:
Thanks for commenting!

I wouldn't say they are missing per se, as there are so many noetic packages and we don't really have a fully automated process of creating AUR packages from the ROS catkin packages.

We could however turn this issue into a package request thread and add the two packages (and their associated deps) to the AUR.

dyamon, 2021-01-10:
Thank, yeah as you can tell I'm new to the world of ROS and it's not even clear to me what these packages do. That said, since they are required for the intro tutorial of a (well established?) tool like MoveIt, I'd say it would be nice to have them.

acxz, 2021-01-10:
> required for the intro tutorial of a (well established?) tool like MoveIt

Agreed

acxz, 2021-01-11:
Done with repos: 

https://github.com/ros-noetic-arch/ros-noetic-moveit-resources-panda-moveit-config

and

https://github.com/ros-noetic-arch/ros-noetic-rviz-visual-tools

closed 2021-01-11 by acxz