ros-noetic-arch/ros-noetic-desktop-full #18: [Package request] moveit-visual-tools, moveit-tutorials

dyamon, 2021-01-11:
As in #17 , the MoveIt tutorial needs the following packages:
- [ ] [ros-noetic-moveit-visual-tools](https://github.com/ros-planning/moveit_visual_tools)
- [ ] [ros-noetic-moveit-tutorials](https://github.com/ros-planning/moveit_tutorials)

Incidentally, do you know the best way to get the whole list of dependeces? Here I'm calling `rosdep` that spits a dependency at a time and the [`moveit.rosinstall` file](https://github.com/ros-planning/moveit/blob/master/moveit.rosinstall) doesn't seem to list all of them.

Thanks again for the help.

acxz, 2021-01-12:
> Incidentally, do you know the best way to get the whole list of dependeces?

Not sure

Do you want to try to make a pull request to add the packages in? I could help you along the way and you can use the previous packages that I just created as a template. Would help lessen the burden in the future and you could also learn a few things!

dyamon, 2021-01-12:
> Do you want to try to make a pull request to add the packages in? I could help you along the way and you can use the previous packages that I just created as a template. Would help lessen the burden in the future and you could also learn a few things!

Wow, that would be super, thanks!
> I have made the repos here that you can send PRs too. Feel free to ask questions, (I don't mind being bugged!). and use the other PKGBUILDs as a template.
> 
> https://github.com/ros-noetic-arch/ros-noetic-moveit-visual-tools
> https://github.com/ros-noetic-arch/ros-noetic-moveit-tutorials

I'll try my best and come back with a first attempt!

dyamon, 2021-01-13:
Ok, I'm getting almost there. I'm going for the `moveit-visual-tools` package first because I'm not sure the other one is really needed. 

There are a few compilation errors, tho. I filed a pull request to hopefully solve some of them.

dyamon, 2021-01-16:
Ok, so there is a pull request for `moveit-visual-tools` as well as an update on `moveit-core` to the latest version. That should be enough to fix all errors I was getting in the "Getting Started" of the MoveIt tutorial.

`moveit-tutorials` shouldn't be needed actually, because the MoveIt tutorial has instruction on how to clone that repo on a catkin workspace locally.

acxz, 2021-01-17:
Sounds good I'll go ahead and close this issue then.

closed 2021-01-17 by acxz