ros-noetic-arch/ros-noetic-desktop-full #19: Failing with new 1.75.0 version of Boost

ckurtz22, 2021-01-19:
I'm not sure where the best place to post this issue is, but I can't install/update ros from the AUR due to the fact that I've updated Boost to 1.75.0, and everything tries to link against the now non existent Boost 1.72.0

acxz, 2021-01-19:
This is a good place to post this issue. The resolution is to rebuild your ros-noetic packages.

ckurtz22, 2021-01-19:
I had tried uninstalling all of my ROS packages and then reinstalling, and it didn't work. I ended up reinstalling my OS cause I had a bunch of other issues occurring, and now after the fact I think it's because I was using yay and it was reinstalling from a cache. Oops.

closed 2021-01-19 by ckurtz22

acxz, 2021-01-19:
Yep there is a difference between reinstall and and rebuild. With yay you can do `yay -Syu ros-noetic-desktop-full --rebuildtree` this will recursively rebuild all packages that depend on ros-noetic-desktop-full. Glad you have it working now tho.