ros-noetic-arch/ros-noetic-desktop-full #21: error making: ros-noetic-rosgraph-msgs

guntursiboro, 2021-04-10:
I am trying to install ros-noetic-desktop-full 1.5.0-1, but terminated because of
`error making: ros-noetic-rosgraph-msgs`

So I spesifically install this package to find out more, I use this command in terminal
`yay -S ros-noetic-rosgraph-msgs --rebuildtree &| tee ros-noetic-rosgraph-msgs.txt`

and here is the output that I get.
[https://gist.github.com/guntursiboro/c8c706744261d8bac861cacdd6c1663c](url)

Thanks in advanced for you help.

AchmadFathoni, 2021-04-22:
Have you fix this?

guntursiboro, 2021-04-22:
How do you mean install manually? from source code? Because all this time I only installed from AUR.

AchmadFathoni, 2021-04-23:
Go to `~/.cache/yay/ros-noetic-genmsg` and then run `makepkg -fiC`

guntursiboro, 2021-04-24:
There is no folder with that name in that location.

guntursiboro, 2021-04-24:
This error, 
` ModuleNotFoundError: No module named 'genmsg'`
Is that related to `ros-noetic-genmsg` package? Because yay didn't seem to download that package.

AchmadFathoni, 2021-04-24:
Yes, this is how I know it: First I source `/opt/ros/noetic/setup.zsh` in my terminal. Second, I run this python program
```
import os
import genmsg
path = os.path.abspath(genmsg.__file__)
print(path)
```
Third, I run `yay -Qo /opt/ros/noetic/lib/python3.9/site-packages/genmsg/__init__.py` in terminal which return `/opt/ros/noetic/lib/python3.9/site-packages/genmsg/__init__.py is owned by ros-noetic-genmsg 0.5.16-2`

AchmadFathoni, 2021-04-28:
> There is no folder with that name in that location.

Maybe the packager forget to include `ros-noetic-genmsg` as dependency for `ros-noetic-rosgraph-msgs`. You can manually install it wtih `yay -S ros-noetic-genmsg --asdeps` and  then reinstall `ros-noetic-desktop-full` .

JonaLoeffler, 2021-05-09:
After the latest updates I am facing the same issue. 

Is it possible the issue comes from the fact that the package is available through both arch4edu and the aur, causing a version mismatch?

AchmadFathoni, 2021-05-09:
Have you tried to reinstall 'ros-noetic-genmsg'?

JonaLoeffler, 2021-05-10:
I fixed my issue by completely uninstalling all ros-noetic-* packages, removing the arch4edu repository and reinstalling using `yay ros-noetic-desktop-full`.

I am no Arch expert, not sure why this helped.

JonaLoeffler, 2021-05-10:
I listed the packages using `yay -Q | grep ros-noetic-` and manually formatted the output to uninstall them all using `yay -R`

AchmadFathoni, 2021-05-10:
> How do you uninstall all ros-noetic-* packages? I tried to uninstall using comman below, but it didn't do anything.
> `sudo pacman -Rsu ros-noetic*`
> It said
> `No matches for wildcard “ros-noetic*”. See `help expand`. sudo pacman -Rsu ros-noetic*`

ROS is AUR package and cannot be installed using pacman. Use yay/other AUR helper
`yay -S ros-noetic-desktop-full`

guntursiboro, 2021-05-10:
> > How do you uninstall all ros-noetic-* packages? I tried to uninstall using comman below, but it didn't do anything.
> > `sudo pacman -Rsu ros-noetic*`
> > It said
> > `No matches for wildcard “ros-noetic*”. See `help expand`. sudo pacman -Rsu ros-noetic*`
> 
> How about `sudo pacman -Rsu ros-noetic-desktop-full`?

Same, not found.  I did what  JonaLoeffler do. 
I am installing now. I hope this will finally installed ros-noetic-full-desktop in my computer.

guntursiboro, 2021-05-10:
> Have you tried to reinstall 'ros-noetic-genmsg'?

This package is not available last time I checked
Sorry for the late reply.

AchmadFathoni, 2021-05-11:
> > Have you tried to reinstall 'ros-noetic-genmsg'?
> 
> This package is not available last time I checked
> Sorry for the late reply. 

https://aur.archlinux.org/packages/ros-noetic-genmsg/

HarrisonDuffield, 2021-05-11:
I got to a point this morning where I had spent too much time  sorting this out for what's it's worth, so yeah friendship ended with Arch, ubuntu now my friend. 

Fundamentally we're looking at a python package not being detected. If it hadn't  installed when I ran the `ros noetic genmsg` there would have been an error, to me that therefore suggests it's installing to the wrong place for the python instance (I believe that should be site-package's), so an environment variable problem.