ros-noetic-arch/ros-noetic-desktop-full #22: Failure while building

EvanLind, 2021-05-26:
==> Making package: ros-noetic-gazebo-ros 2.9.1-1 (Wed 26 May 2021 01:23:05 PM CST)

```c++
[ 40%] Building CXX object CMakeFiles/gazebo_ros_api_plugin.dir/src/gazebo_ros_api_plugin.cpp.o
In file included from /usr/include/ignition/math6/ignition/math/Quaternion.hh:20,
                 from /usr/include/ignition/math6/ignition/math/Pose3.hh:20,
                 from /usr/include/sdformat-9.5/sdf/Actor.hh:23,
                 from /usr/include/sdformat-9.5/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Events.hh:21,
                 from /home/evanlin/.cache/yay/ros-noetic-gazebo-ros/src/gazebo_ros_pkgs-2.9.1/gazebo_ros/src/gazebo_ros_api_plugin.cpp:23:
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::duration ignition::math::v6::stringToDuration(const string&)’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:990:28: error: ‘chrono_literals’ is not a namespace-name
  990 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::time_point ignition::math::v6::stringToTimePoint(const string&)’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:1032:28: error: ‘chrono_literals’ is not a namespace-name
 1032 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:56: error: unable to find numeric literal operator ‘operator""s’
 1033 |       std::chrono::steady_clock::time_point timePoint{-1s};
      |                                                        ^~
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:56: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:58: error: no matching function for call to ‘std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(<brace-enclosed initializer list>)’
 1033 |       std::chrono::steady_clock::time_point timePoint{-1s};
```

And I have tried `yay -Syu ros-noetic-desktop-full --rebuildtree`, but it still doesn't work

AchmadFathoni, 2021-05-26:
it should be in `/usr/include/c++/11.1.0/chrono` line 3165
```
inline namespace chrono_literals
    {
  #pragma GCC diagnostic push
  #pragma GCC diagnostic ignored "-Wliteral-suffix"
```
Are you sure that you install gcc correctly? Mine is gcc 11.1.0-1

EvanLind, 2021-05-26:
[image]
```c++
~ ❯❯❯ gcc --version 
gcc (GCC) 11.1.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
The gcc version is correct, ‘chrono_literals’ namespace-name is also defined...

AchmadFathoni, 2021-05-26:
Hm, What is the version of your ignition-math? run this `yay -Qi ignition-math`

EvanLind, 2021-05-26:
https://aur.archlinux.org/cgit/aur.git/log/?h=ignition-math
newest 6.8.0-1, I try to downgrade to 6.7.0 and rebuild, but it still doesn't work
```c++
~ ❯❯❯ yay -Qi ignition-math
Name            : ignition-math
Version         : 6.8.0-1
Description     : Math classes and functions for robot applications
Architecture    : x86_64
URL             : https://ignitionrobotics.org/libs/math
Licenses        : Apache
Groups          : development
Provides        : None
Depends On      : gcc-libs
Optional Deps   : eigen [installed]
Required By     : gazebo  ignition-common  ignition-common-3  ignition-msgs-5  sdformat-9
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 2.71 MiB
Packager        : Unknown Packager
Build Date      : Wed 26 May 2021 12:48:49 PM CST
Install Date    : Wed 26 May 2021 12:51:53 PM CST
Install Reason  : Explicitly installed
Install Script  : No
Validated By    : None
```

AchmadFathoni, 2021-05-26:
Ok I I think I found the the problem, the `ros-noetic-gazebo-ros` in AUR is outdated. I am also failed to build 2.9.1 with same error. Current version that can be build is 2.9.2.

These are the oudated AUR packages

ros-noetic-controller-manager-msgs
ros-noetic-rqt-srv
ros-noetic-rqt-reconfigure
ros-noetic-rospy
ros-noetic-roscpp
ros-noetic-gazebo-ros-pkgs
ros-noetic-gazebo-ros-control
ros-noetic-gazebo-ros
ros-noetic-gazebo-plugins
ros-noetic-gazebo-dev
ros-noetic-gazebo-msgs
ros-noetic-gazebo-dev

You can find updated version in my github repo for example `https://github.com/AchmadFathoni/ros-noetic-gazebo-ros`

EvanLind, 2021-05-26:
`makepkg` for each package manually? 
ros-noetic-controller-manager-msgs  error for :
```
[  1%] Generating Lisp code from controller_manager_msgs/ControllerState.msg
Traceback (most recent call last):
  File "/opt/ros/noetic/share/genlisp/cmake/../../../lib/genlisp/gen_lisp.py", line 39, in <module>
    import genlisp
ModuleNotFoundError: No module named 'genlisp'
```
```
pip install genlisp                                                                                                              ✘ 4 master ◼
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement genlisp (from versions: none)
ERROR: No matching distribution found for genlisp
```
Could you please update to AUR official repo?

AchmadFathoni, 2021-05-26:
For `ModuleNotFoundError: No module named 'genlisp'`, just install `ros-noetic-genlisp`. If you support more updated AUR, you can complain at #20 and [here](https://bbs.archlinux.org/viewtopic.php?id=266644)

AchmadFathoni, 2021-05-26:
Remove the src folder and build again

EvanLind, 2021-05-26:
I have already rm folder src, and run `makepkg -fiC`, still the same error
`/usr/include/ignition/math6/ignition/math/Helpers.hh:990:28: error: ‘chrono_literals’ is not a namespace-name`
Do you build ros-noetic-gazebo-ros, ros-noetic-gazebo-plugins successfully for version 2.9.2

AchmadFathoni, 2021-05-26:
Oh, I am sorry. After I delete my own src and do clean build, I get your error. The problem is: after newest gcc update ros package needs to explicitly use `-DCMAKE_CXX_STANDARD=17` flag in PKGBUILD. I will update my repo, maybe 10 minute or so.

AchmadFathoni, 2021-05-26:
ros-noetic-gazebo-ros is updated

AchmadFathoni, 2021-05-26:
and ros-noetic-gazebo-plugins

AchmadFathoni, 2021-05-26:
If you done please close this issue @EvanLind

EvanLind, 2021-05-26:
Build successfully after I add `-DCMAKE_CXX_STANDARD=17` to AUR 2.9.1 ros-noetic-gazebo-ros PKGBUILD

EvanLind, 2021-05-26:
`using namespace` is supported since C++11, why must flag `-DCMAKE_CXX_STANDARD=17` be explicitly add in cmake? How did you discover the solution to the problem

AchmadFathoni, 2021-05-26:
Why C++17? Because it's work+newer and time always moving forward.

How do I know that flag solved the problem? So, there is amazing technology called search engine, the most popular one is Google. You can type your error there and hit enter and then a catalog of link will be displayed. You click one of the link in that catalog and you will be redirected to a website that contain information about the solution candidate. If the first solution candidate doesn't work, you can try the next link in catalog.

EvanLind, 2021-05-26:
I hace tried several solution candidates with "amazing google", but it doesn't work... Then I found ignition-math update 4 days ago, misunderstood it was a version update problem.
Thanks all the same.