ros-noetic-arch/ros-noetic-desktop-full #25: Unable to build packages

KSaiAkshit, 2021-08-05:
I have recently switched over from Ubuntu 20.04 to Arch (ArcoLinux) and i am trying to get ROS working on Arch but to no avail.
I followed the instructions on [this page](http://wiki.ros.org/noetic/Installation/ArchLinux#Arch_Linux_install_of_ROS_Noetic), but keep running into errors.
I've tried building it via Paru but i get the same error of unmet dependencies.
I have added arch4edu to my /etc/pacman.conf and updated it.
None of the other "ros bundles" work either.

I'd greatly appreciate if someone could help me

AchmadFathoni, 2021-08-05:
I heard people got trouble from arch4edu. Are you really need arch4edu?

KSaiAkshit, 2021-08-05:
> I heard people got trouble from arch4edu. Are you really need arch4edu?

I don't really know, the official Ros-wiki page does mention it and i did add the repo.

AchmadFathoni, 2021-08-05:
Ok, first. I want you to remove your arch4edu from your arch. I never add third party repo such as arch4edu so you must find how to do it yourself. Google it!

KSaiAkshit, 2021-08-05:
> Ok, first. I want you to remove your arch4edu from your arch. I never add third party repo such as arch4edu so you must find how to do it yourself. Google it!

I removed it from my pacman.conf

AchmadFathoni, 2021-08-05:
Now try install ros-noetic-desktop-full again using paru. Tell me the output.

KSaiAkshit, 2021-08-05:
> Now try install ros-noetic-desktop-full again using paru. Tell me the output.

```bash
/opt/ros/noetic/share/genmsg/cmake/pkg-genmsg.cmake.em:56: error: <class 'ModuleNotFoundError'>: No module named 'genmsg'
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/em.py", line 3303, in <module>
    if __name__ == '__main__': main()
  File "/usr/lib/python3.9/site-packages/em.py", line 3301, in main
    invoke(sys.argv[1:])
  File "/usr/lib/python3.9/site-packages/em.py", line 3284, in invoke
    interpreter.wrap(interpreter.file, (file, name))
  File "/usr/lib/python3.9/site-packages/em.py", line 2295, in wrap
    self.fail(e)
  File "/usr/lib/python3.9/site-packages/em.py", line 2284, in wrap
    callable(*args)
  File "/usr/lib/python3.9/site-packages/em.py", line 2359, in file
    self.safe(scanner, done, locals)
  File "/usr/lib/python3.9/site-packages/em.py", line 2401, in safe
    self.parse(scanner, locals)
  File "/usr/lib/python3.9/site-packages/em.py", line 2421, in parse
    token.run(self, locals)
  File "/usr/lib/python3.9/site-packages/em.py", line 1425, in run
    interpreter.execute(self.code, locals)
  File "/usr/lib/python3.9/site-packages/em.py", line 2596, in execute
    _exec(statements, self.globals, locals)
  File "<string>", line 5, in <module>
ModuleNotFoundError: No module named 'genmsg'
CMake Error at /opt/ros/noetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):

  execute_process(/home/akshit/.cache/yay/ros-noetic-rosgraph-msgs/src/build/catkin_generated/env_cached.sh
  "/usr/bin/python" "/usr/lib/python3.9/site-packages/em.py" "--raw-errors"
  "-F"
  "/home/akshit/.cache/yay/ros-noetic-rosgraph-msgs/src/build/cmake/rosgraph_msgs-genmsg-context.py"
  "-o"
  "/home/akshit/.cache/yay/ros-noetic-rosgraph-msgs/src/build/cmake/rosgraph_msgs-genmsg.cmake"
  "/opt/ros/noetic/share/genmsg/cmake/pkg-genmsg.cmake.em") returned error
  code 1
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/em_expand.cmake:25 (safe_execute_process)
  /opt/ros/noetic/share/genmsg/cmake/genmsg-extras.cmake:303 (em_expand)
  CMakeLists.txt:12 (generate_messages)

-- Configuring incomplete, errors occurred!
See also "/home/akshit/.cache/yay/ros-noetic-rosgraph-msgs/src/build/CMakeFiles/CMakeOutput.log".
See also "/home/akshit/.cache/yay/ros-noetic-rosgraph-msgs/src/build/CMakeFiles/CMakeError.log".
==> ERROR: A failure occurred in build().
    Aborting...
error making: ros-noetic-rosgraph-msgs
```

This is what i get when ran using yay without arch4edu

AchmadFathoni, 2021-08-05:
Yeah this is the exact error people get when installing arch4edu. Now install  genmsg by `yay -S ros-noetic-genmsg`

KSaiAkshit, 2021-08-05:
> Yeah this is the exact error people get when installing arch4edu. Now install genmsg by `yay -S ros-noetic-genmsg`

I had already removed arch4edu from pacman.conf, but I could install genmsg
```bash❯ yay -S ros-noetic-genmsg
:: Checking for conflicts...
:: Checking for inner conflicts...
[Aur:1]  ros-noetic-genmsg-0.5.16-2

:: Downloaded PKGBUILD (1/1): ros-noetic-genmsg
  1 ros-noetic-genmsg                (Installed) (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> n
:: (1/1) Parsing SRCINFO: ros-noetic-genmsg
==> Making package: ros-noetic-genmsg 0.5.16-2 (Thu 05 Aug 2021 05:29:27 PM IST)
==> Retrieving sources...
  -> Downloading ros-noetic-genmsg-0.5.16.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   118  100   118    0     0    223      0 --:--:-- --:--:-- --:--:--   223
100 60357    0 60357    0     0  42869      0 --:--:--  0:00:01 --:--:--  118k
==> Validating source files with sha256sums...
    ros-noetic-genmsg-0.5.16.tar.gz ... Passed
==> Making package: ros-noetic-genmsg 0.5.16-2 (Thu 05 Aug 2021 05:29:29 PM IST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found ros-noetic-genmsg-0.5.16.tar.gz
==> Validating source files with sha256sums...
    ros-noetic-genmsg-0.5.16.tar.gz ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting ros-noetic-genmsg-0.5.16.tar.gz with bsdtar
==> Sources are ready.
==> Making package: ros-noetic-genmsg 0.5.16-2 (Thu 05 Aug 2021 05:29:31 PM IST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Starting build()...
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/akshit/.cache/yay/ros-noetic-genmsg/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python (found suitable version "3.9.6", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.9/site-packages/em.py
-- Using empy: /usr/lib/python3.9/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /home/akshit/.cache/yay/ros-noetic-genmsg/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found Python: /usr/bin/python3.9 (found version "3.9.6") found components: Interpreter
-- Found Threads: TRUE
-- Using Python nosetests: /usr/bin/nosetests-3.9
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
[... 87 lines cut ...]
[sudo] password for akshit:
loading packages...
warning: ros-noetic-genmsg-0.5.16-2 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Package (1)        Old Version  New Version  Net Change

ros-noetic-genmsg  0.5.16-2     0.5.16-2       0.00 MiB

Total Installed Size:  0.14 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n]
(1/1) checking keys in keyring                                                                                                                 [---------------------------------------------------------------------------------------] 100%
(1/1) checking package integrity                                                                                                               [---------------------------------------------------------------------------------------] 100%
(1/1) loading package files                                                                                                                    [---------------------------------------------------------------------------------------] 100%
(1/1) checking for file conflicts                                                                                                              [---------------------------------------------------------------------------------------] 100%
(1/1) checking available disk space                                                                                                            [---------------------------------------------------------------------------------------] 100%
:: Processing package changes...
(1/1) reinstalling ros-noetic-genmsg                                                                                                           [---------------------------------------------------------------------------------------] 100%
:: Running post-transaction hooks...
(1/1) Refreshing PackageKit...
```
Installed sucessfully!

AchmadFathoni, 2021-08-05:
Now try open python in terminal then type `ìmport genmsg`

AchmadFathoni, 2021-08-05:
Ok, to be safe I think you should clear your cache by running `rm -rf ~/.cache/yay/ros-noetic-*`

KSaiAkshit, 2021-08-05:
> Now try open python in terminal then type `ìmport genmsg`

```bash
python
Python 3.9.6 (default, Jun 30 2021, 10:22:16)
[GCC 11.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import genmsg
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'genmsg'
```
Module not found

AchmadFathoni, 2021-08-05:
Is `/opt/ros/noetic/lib/python3.9/site-packages/genmsg/__init__.py` exist?

KSaiAkshit, 2021-08-05:
> Is `/opt/ros/noetic/lib/python3.9/site-packages/genmsg/__init__.py` exist?

yes

AchmadFathoni, 2021-08-05:
what is the output of `echo $PYTHONPATH`?

KSaiAkshit, 2021-08-05:
> what is the output of `echo $PYTHONPATH`?

its Empty

AchmadFathoni, 2021-08-05:
Try to run `source /opt/ros/noetic/setup.bash` and do `import genmsg` in terminal python again.

KSaiAkshit, 2021-08-05:
> Try to run `source /opt/ros/noetic/setup.bash` and do `import genmsg` in terminal python again.

Works, I am able to import genmsg

AchmadFathoni, 2021-08-05:
You are using yay right? What is the content of '~/.cache/yay/ros-noetic-graph-msgs/PKGBUILD'?

KSaiAkshit, 2021-08-05:
> You are using yay right? What is the content of '~/.cache/yay/ros-noetic-graph-msgs/PKGBUILD'?

You had me clear cache:-

> Ok, to be safe I think you should clear your cache by running `rm -rf ~/.cache/yay/ros-noetic-*`

AchmadFathoni, 2021-08-05:
Ok, please try to install ros-noetic-rosgraph-msgs again and if fail send the PKGBUILD here

KSaiAkshit, 2021-08-05:
```bash
# Script generated with import_catkin_packages.py.
# For more information: https://github.com/bchretien/arch-ros-stacks.
pkgdesc="ROS - Messages relating to the ROS Computation Graph."
url='https://wiki.ros.org/rosgraph_msgs'

pkgname='ros-noetic-rosgraph-msgs'
pkgver='1.11.3'
arch=('any')
pkgrel=1
license=('BSD')

ros_makedepends=(
	ros-noetic-message-generation
	ros-noetic-std-msgs
	ros-noetic-catkin
)

makedepends=(
	'cmake'
	'ros-build-tools'
	${ros_makedepends[@]}
)

ros_depends=(
	ros-noetic-message-runtime
	ros-noetic-std-msgs
)

depends=(
	${ros_depends[@]}
)

_dir="ros_comm_msgs-${pkgver}/rosgraph_msgs"
source=("${pkgname}-${pkgver}.tar.gz"::"https://github.com/ros/ros_comm_msgs/archive/${pkgver}.tar.gz")
sha256sums=('5b8b91e8671d03ea84ba32a3ea7360bc4594655e7ba3ec6677a984f393aaafbd')

build() {
	# Use ROS environment variables.
	source /usr/share/ros-build-tools/clear-ros-env.sh
	[ -f /opt/ros/noetic/setup.bash ] && source /opt/ros/noetic/setup.bash

	# Create the build directory.
	[ -d ${srcdir}/build ] || mkdir ${srcdir}/build
	cd ${srcdir}/build

	# Build the project.
	cmake ${srcdir}/${_dir} \
		-DCATKIN_BUILD_BINARY_PACKAGE=ON \
		-DCMAKE_INSTALL_PREFIX=/opt/ros/noetic \
		-DPYTHON_EXECUTABLE=/usr/bin/python \
		-DSETUPTOOLS_DEB_LAYOUT=OFF
	make
}

package() {
	cd "${srcdir}/build"
	make DESTDIR="${pkgdir}/" install
}
```
I couldnt upload the File, github apparently doesnt support PKGBUILD

AchmadFathoni, 2021-08-05:
> ```bash
> # Script generated with import_catkin_packages.py.
> # For more information: https://github.com/bchretien/arch-ros-stacks.
> pkgdesc="ROS - Messages relating to the ROS Computation Graph." [...]

Do you have same error as before when installing this package?

KSaiAkshit, 2021-08-05:
> If you still have same error then uninstall every ros-noetic-* packages from your arch then try to install ros-noetic-rosgraph-msgs again.
> 
> 
> If you still have same error try to reinstall your arch linux then install yay then install ros-noetic-desktop-full again. DO NOT use arch4edu

How do I uninstall all of them?
pacman -R ros-noetic-*.  ?

KSaiAkshit, 2021-08-05:
> How do I uninstall all of them?
> `pacman -R ros-noetic-*` ?

I can't use `sudo pacman -Rsc ros-melodic-*` , it returns package not found.

AchmadFathoni, 2021-08-05:
`yay -Rs $(yay -Qm | grep ros | cut -f 1 -d " ")`

AchmadFathoni, 2021-08-05:
Then `yay -Yc`

KSaiAkshit, 2021-08-05:
> `yay -Rs $(yay -Qm | grep ros | cut -f 1 -d " ")`

`sudo pacman -Rsc $(pacman -Qs ros- | awk ' NR % 2 == 1 { print substr($1, 7) } ')`
i found this on AUR page

KSaiAkshit, 2021-08-05:
> Then `yay -Yc`

This will removing a lot of stuff that have nothing to do with ROS, so i didnt proceed with this.

AchmadFathoni, 2021-08-05:
It removes unused dependency. Yeah you may do that, tell me when you success building ros-noetic-desktop-full

KSaiAkshit, 2021-08-05:
> It removes unused dependency. Yeah you may do that, tell me when you success building ros-noetic-desktop-full

What do i do after removing said stuff?
`yay -S ros-noetic-desktop-full` ?

AchmadFathoni, 2021-08-05:
Yes

AchmadFathoni, 2021-08-05:
Does it work?

KSaiAkshit, 2021-08-07:
> Does it work?

Nope, not happening :cry:

AchmadFathoni, 2021-08-07:
Yeah maybe just reinstall your arch, or use Ubuntu instead

KSaiAkshit, 2021-08-07:
> Yeah maybe just reinstall your arch, or use Ubuntu instead

I have switched over from Ubuntu actually 😅

closed 2021-08-07 by KSaiAkshit

AchmadFathoni, 2021-08-07:
Well, I've tried to install ros via arch4edu. I got the same error as above. The fix is actually simple:
1. Delete arch4edu from pacman.conf
2. Delete `~/.cache/yay/`
3. Delete all ros package and its dependency `yay -Rs $(yay -Q |grep ros- | cut -d " " -f1 && yay -Yc`
4. Update database `yay -Syyu`
5. Install ros `yay -S ros-noetic-desktop-full --noconfirm --sudoloop`
@KSaiAkshit  seem you miss the step 4