ros-noetic-arch/ros-noetic-desktop-full #28: ros-noetic-rviz build failed

RaviRahar, 2021-10-12:
Getting following error while building: 

``` ==> Starting build()...
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: /usr/lib64/cmake/Boost-1.76.0/BoostConfig.cmake (found version "1.76.0") found components: filesystem program_options system thread
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.0")
-- Checking for modules 'OGRE;OGRE-Overlay'
--   Found OGRE, version 1.9.0
--   Found OGRE-Overlay, version 1.9.0
-- OGRE_INCLUDE_DIRS=/usr/include/OGRE;/usr/include/OGRE/Overlay
-- OGRE_LIBRARIES=/usr/lib/libOgreOverlay.so;/usr/lib/libOgreMain.so;/usr/lib/libpthread.so
-- OGRE_PLUGIN_DIR=/usr/lib/OGRE
-- Found OpenGL: /usr/lib/libOpenGL.so
-- Using CATKIN_DEVEL_PREFIX: /home/uroborosikes/.cache/yay/ros-noetic-rviz/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python (found suitable version "3.9.7", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.9/site-packages/em.py
-- Using empy: /usr/lib/python3.9/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /home/uroborosikes/.cache/yay/ros-noetic-rviz/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found Python: /usr/bin/python3.9 (found version "3.9.7") found components: Interpreter
-- Found Threads: TRUE
-- Using Python nosetests: /usr/bin/nosetests-3.9
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /opt/ros/noetic/share/urdf/cmake/urdfConfig.cmake:173 (message):
  Project 'rviz' tried to find library
  '$<$<NOT:$<CONFIG:Debug>>:/usr/lib/liburdfdom_sensor.so>'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'urdf'? Did you find_package() it before the subdirectory containing its
  code is included?
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:117 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/uroborosikes/.cache/yay/ros-noetic-rviz/src/build/CMakeFiles/CMakeOutput.log".
See also "/home/uroborosikes/.cache/yay/ros-noetic-rviz/src/build/CMakeFiles/CMakeError.log".
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ros-noetic-rviz 
```

Tried Clean Build of all packages with --rebuildtree
- `yay -Rns $(yay -Qsq ros-noetic`
- `yay -S ros-noetic-desktop-full --rebuildtree`

How can I resolve this issue?

AchmadFathoni, 2021-10-12:
And blacklist it from being updated

RaviRahar, 2021-10-12:
Can you please guide me how do I do it?
Current urdfdom version is 3.0.0-1 which is dependency of ros-noetic-urdf-1.13.2-2
So I removed it ` sudo pacman -Rns ros-noetic-urdf `
what do I install now? which version of ros-noetic-urdf?
And if I ignore urdf from updating, what to do with ros-noetic-urdf?

AchmadFathoni, 2021-10-12:
Ok
1. Install the ros-noetic-urdf back by `yay -S ros-noetic-urdf`
2.  `cd ~/.cache/yay/urdfdom`
3. `git checkout master~`
4. `makepkg -fi`

AchmadFathoni, 2021-10-12:
Then put 
`IgnorePkg   = urdfdom`
in your /etc/pacman.conf
You should be able to install ROS again

RaviRahar, 2021-10-18:
It probably is.
My laptop is old, so the build stuck at 648/1100+ (it stuck twice before at 500 something and was taking multiple hrs. per build). And our prof. made it necessary to install ubuntu. 
But after installing the packages you mentioned, I think it probably would have installed.
I will close the issue, will test back in couple of months with another laptop.

closed 2021-10-18 by RaviRahar