ros-noetic-arch/ros-noetic-desktop-full #29: Problem building ros-noetic-qt-gui-cpp-0.4.2

nosense345, 2021-11-05:
Is there a way around this error? 
I can't find much info.

Running SIP generator for qt_gui_cpp_sip Python bindings...
sip: Deprecation warning: qt_gui_cpp.sip:1: %Module version number should be specified using the 'version' argument
sip: /usr/lib/python3.9/site-packages/PyQt5/bindings/QtCore/QtCoremod.sip:23: syntax error
Traceback (most recent call last):
  File "/opt/ros/noetic/share/python_qt_binding/cmake/sip_configure.py", line 122, in <module>
    subprocess.check_call(cmd)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', '/home/klaas/.cache/yay/ros-noetic-qt-gui-cpp/src/build/sip/qt_gui_cpp_sip', '-b', '/home/klaas/.cache/yay/ros-noetic-qt-gui-cpp/src/build/sip/qt_gui_cpp_sip/pyqtscripting.sbf', '-I', '/usr/lib/python3.9/site-packages/PyQt5/bindings', '-w', '-n', 'PyQt5.sip', '-t', 'Qt_5_15_0', '-t', 'WS_X11', 'qt_gui_cpp.sip']' returned non-zero exit status 1.
make[2]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/build.make:103: sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [CMakeFiles/Makefile2:478: src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().

this is the offending line:
%Module(name=PyQt5.QtCore, call_super_init=True, default_VirtualErrorHandler=PyQt5, keyword_arguments="Optional", use_limited_api=True,  py_ssize_t_clean=True)

versions:
extra/python-pyqt5-sip 12.9.0-1
extra/python-sip4 4.19.25-1
extra/sip4 4.19.25-1

thank you very much for your help.
Klaas

FIXMBR, 2021-11-06:
I'm dealing with similar error while trying to rebuild rviz.
```
sip: /usr/lib/python3.9/site-packages/PyQt5/bindings/QtCore/QtCoremod.sip:23: syntax error
Traceback (most recent call last):
  File "/opt/ros/noetic/share/python_qt_binding/cmake/sip_configure.py", line 122, in <module>
    subprocess.check_call(cmd)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', '/home/stefe/.cache/yay/ros-noetic-rviz/src/build/sip/rviz_sip', '-b', '/home/stefe/.cache/yay/ros-noetic-rviz/src/build/sip/rviz_sip/pyqtscripting.sbf', '-I', '/usr/lib/python3.9/site-packages/PyQt5/bindings', '-w', '-n', 'PyQt5.sip', '-t', 'Qt_5_15_0', '-t', 'WS_X11', 'rviz.sip']' returned non-zero exit status 1.
make[2]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:108: sip/rviz_sip/Makefile] Error 11
make[1]: *** [CMakeFiles/Makefile2:4831: src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2
```

AchmadFathoni, 2021-11-06:
The fix is to upgrade sip v4 to v4.19.25, [source](https://github.com/ros-visualization/qt_gui_core/issues/249)

AchmadFathoni, 2021-11-06:
Wait, in Arch it already 4.19.25

FIXMBR, 2021-11-07:
After running system update today it worked. I think the fix was updating `python-pyqt5 5.15.6-1` to `python-pyqt5 5.15.6-2`.

nosense345, 2021-11-10:
Seems to work for me too: ros-noetic-qt-gui compiles ok now.

Thank you.

closed 2021-11-10 by AchmadFathoni