ros-noetic-arch/ros-noetic-desktop-full #30: ros-noetic-pcl-ros

nosense345, 2021-11-10:
I have run into a new challenge.
ros-noetic-pcl-ros isn't compiling:

-- Could not find the required component 'dynamic_reconfigure'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "dynamic_reconfigure" with any of the following names:

    dynamic_reconfigureConfig.cmake
    dynamic_reconfigure-config.cmake

  Add the installation prefix of "dynamic_reconfigure" to CMAKE_PREFIX_PATH
  or set "dynamic_reconfigure_DIR" to a directory containing one of the above
  files.  If "dynamic_reconfigure" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:39 (find_package)

-- Configuring incomplete, errors occurred!

I have manually installed the dynamic_reconfigure package as found on AUR (ros-noetic-dynamic-reconfigure 1.7.1-1) and this package seems to install Ok, as in  
/opt/ros/noetic/share/dynamic_reconfigure/cmake/dynamic_reconfigureConfig.cmake
exists 
but that doesn't seem to help.

The log file seems to suggest pthread is involved somehow:

make[1]: Entering directory '/home/klaas/.cache/yay/ros-noetic-pcl-ros/src/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_2fc77.dir/src.c.o
/usr/bin/cc -DCMAKE_HAVE_LIBC_PTHREAD  -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIF
Y_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection  -o CMakeFiles/cmTC_2fc77.di
r/src.c.o -c /home/klaas/.cache/yay/ros-noetic-pcl-ros/src/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_2fc77
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_2fc77.dir/link.txt --verbose=1
/usr/bin/cc -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror
=format-security         -fstack-clash-protection -fcf-protection  -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now  CMa
keFiles/cmTC_2fc77.dir/src.c.o -o cmTC_2fc77 
/usr/bin/ld: CMakeFiles/cmTC_2fc77.dir/src.c.o: in function `main':
src.c:(.text.startup+0x2a): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text.startup+0x34): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text.startup+0x3e): undefined reference to `pthread_cancel'
/usr/bin/ld: src.c:(.text.startup+0x4a): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_2fc77.dir/build.make:99: cmTC_2fc77] Error 1
make[1]: Leaving directory '/home/klaas/.cache/yay/ros-noetic-pcl-ros/src/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:127: cmTC_2fc77/fast] Error 2

Source file was:
#include <pthread.h>

static void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Any suggestions on how to fix this?

Thank you very much for your help.

AchmadFathoni, 2021-11-10:
Sorry but I can't reproduce this

nosense345, 2021-11-11:
So: start from scratch.
Re-install all installed standard packages.
Do the --rebuildtree thing for ros-noetic for all packages.
that gets me stuck on  python-catkin_pkg 0.4.24-1 which was fine before I started from scratch

copying src/catkin_pkg/templates/metapackage.cmake.in -> build/lib/catkin_pkg/templates
Traceback (most recent call last):
  File "/home/klaas/.cache/yay/python-catkin_pkg/src/catkin_pkg-0.4.24/setup.py", line 58, in <module>
    setup(**kwargs)
  File "/usr/lib/python3.9/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.9/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.9/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.9/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/usr/lib/python3.9/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.9/site-packages/setuptools/command/build_py.py", line 56, in run
    self.build_package_data()
  File "/usr/lib/python3.9/site-packages/setuptools/command/build_py.py", line 127, in build_package_data
    srcfile in self.distribution.convert_2to3_doctests):
AttributeError: 'Distribution' object has no attribute 'convert_2to3_doctests'

Which makes me wonder:  what is the trick to get a full set of buildable packages on a rolling release for something as big as ros-noetic?

AchmadFathoni, 2021-11-11:
Cannot duplicate that neither, try reinstall your Arch, install yay, install ros-noetic-desktop-full

AchmadFathoni, 2021-11-18:
Damn, something in pcl-ros'S CMakeLists.txt erase the CMAKE_INSTALL_PREFIX environment variable. This is weird. Anyway I've push the fix using explicit cmake argument.

AchmadFathoni, 2021-11-19:
Upstream issue at https://github.com/ros-perception/perception_pcl/issues/351

AchmadFathoni, 2021-12-21:
@nosense345  The patch is out. First, you must rebuild the pcl package with this patch
```
$ cd .cache/yay/pcl
$ git am -s < <(curl -s http://ix.io/3IRq)
$ makepkg -fiCc
```
Then rebuild the ros-noetic-pcl-ros
```
$ cd ../ros-noetic-pcl-ros
$ makepkg -fiCc
```