ros-noetic-arch/ros-noetic-desktop-full #31: Could not find tf2_geometry_msgs

blegat, 2021-11-13:
It seems https://aur.archlinux.org/packages/ros-noetic-tf2-geometry-msgs/ should be added as a dependency. Trying to install ros-noetic-desktop-full, I got:
```
Packages (1) ros-noetic-tf-1.13.2-2

Total Installed Size:  1.05 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking keys in keyring                                                                          [##############################################################] 100%
(1/1) checking package integrity                                                                        [##############################################################] 100%
(1/1) loading package files                                                                             [##############################################################] 100%
(1/1) checking for file conflicts                                                                       [##############################################################] 100%
(1/1) checking available disk space                                                                     [##############################################################] 100%
:: Processing package changes...
(1/1) installing ros-noetic-tf                                                                          [##############################################################] 100%
==> Making package: ros-noetic-laser-geometry 1.6.7-2 (Sat 13 Nov 2021 05:39:44 PM EST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Downloading ros-noetic-laser-geometry-1.6.7.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   136  100   136    0     0   1228      0 --:--:-- --:--:-- --:--:--  1236
100 19095    0 19095    0     0  70867      0 --:--:-- --:--:-- --:--:-- 70867
==> Validating source files with sha256sums...
    ros-noetic-laser-geometry-1.6.7.tar.gz ... Passed
==> Extracting sources...
  -> Extracting ros-noetic-laser-geometry-1.6.7.tar.gz with bsdtar
==> Starting prepare()...
==> Starting build()...
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/blegat/.cache/aurman/ros-noetic-laser-geometry/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python (found suitable version "3.9.7", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.9/site-packages/em.py  
-- Using empy: /usr/lib/python3.9/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /home/blegat/.cache/aurman/ros-noetic-laser-geometry/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found Python: /usr/bin/python3.9 (found version "3.9.7") found components: Interpreter 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-3.9
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Could NOT find tf2_geometry_msgs (missing: tf2_geometry_msgs_DIR)
-- Could not find the required component 'tf2_geometry_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "tf2_geometry_msgs"
  with any of the following names:

    tf2_geometry_msgsConfig.cmake
    tf2_geometry_msgs-config.cmake

  Add the installation prefix of "tf2_geometry_msgs" to CMAKE_PREFIX_PATH or
  set "tf2_geometry_msgs_DIR" to a directory containing one of the above
  files.  If "tf2_geometry_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/blegat/.cache/aurman/ros-noetic-laser-geometry/src/build/CMakeFiles/CMakeOutput.log".
See also "/home/blegat/.cache/aurman/ros-noetic-laser-geometry/src/build/CMakeFiles/CMakeError.log".
```

AchmadFathoni, 2021-11-14:
Here is the summary of ros-noetic-laser-geometry dependency produced by `pactree`
```
ros-noetic-laser-geometry
├─ros-noetic-tf
│ ├─ros-noetic-tf2-ros
│ │ ├─ros-noetic-tf2-msgs
```
If pactree can resolve it, your AUR helper should be able too

AchmadFathoni, 2021-12-21:
Sorry, I misread your text. Dependecy now fixed in https://github.com/ros-noetic-arch/ros-noetic-laser-geometry/commit/4f274be739d0d3bfde588d3f30d1a7bcfcae36f5

closed 2021-12-21 by AchmadFathoni