ros-noetic-arch/ros-noetic-desktop-full #32: Support in official repository?

ShadowWarfare, 2021-11-30:
As stated in the AUR comment which I did at https://aur.archlinux.org/packages/ros-noetic-desktop-full/#news - could ROS support be added in the official repository and maintained in the official repository like [extra], [community] for example. It would be nice if Arch supported ROS fully and officially.

AchmadFathoni, 2021-12-01:
I support it, but how do we convince Arch maintainer?

ShadowWarfare, 2021-12-01:
I don't know personally I don't even know where to begin but with Arch supporting things like kubernetes, docker, tensorflow etc. I thought to myself why not ROS as well.

ShadowWarfare, 2021-12-09:
Well I now know that its not possible in the Arch's bug tracker because I did post a package request aka feature request for that package to be placed in the official repository however it was closed straight away and the person who closed it stated that the bug tracker is not the place for package requests. So I really don't know where else to convince a maintainer to support it officially. See https://bugs.archlinux.org/task/72955

AchmadFathoni, 2021-12-09:
Well, we should ask arch forum or arch subreddit where to request official package.

AchmadFathoni, 2021-12-21:
Maybe [this](https://wiki.archlinux.org/title/AUR_Trusted_User_Guidelines) will help

AchmadFathoni, 2021-12-21:
You need to be a Trusted User to adopt an AUR package to Community repo

acxz, 2022-01-31:
This is a great idea and the ultimate goal of this project. However, there is still a lot of work required to make these packages ready for a release in [community]. Before that step, we are aiming to deploy the `ros-noetic-desktop-full` to [arch4edu] as doing so will help us discover build issues that would be present for Trusted Users if these packages are deployed to [community] as is.

The main bottleneck right now, is that many packages do not yet build in a clean chroot. I personally believe that this goal is so far out that working on ros2 for archlinux is more realistic as by the time the ros-noetic packages are cleaned up, ros-noetic might have become much less popular.

closed 2022-01-31 by acxz