ros-noetic-arch/ros-noetic-desktop-full #33: Error while building ros-noetic-rviz

prirai, 2021-12-17:
```
/opt/ros/noetic/share/python_qt_binding/cmake/sip_configure.py:2: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.spawn import find_executable
sip: /usr/lib/python3.10/site-packages/PyQt5/bindings/QtCore/QtCoremod.sip:23: syntax error
Traceback (most recent call last):
  File "/opt/ros/noetic/share/python_qt_binding/cmake/sip_configure.py", line 122, in <module>
    subprocess.check_call(cmd)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', '/home/me/.cache/yay/ros-
noetic-rviz/src/build/sip/rviz_sip', '-b', '/home/me/.cache/yay/ros-noetic-rviz/src/build/sip/rviz_sip/pyqtscripting.sbf', '-I', '/usr/lib/python3.10/site-packages/PyQt5/bindings', '-w', '-n', 'PyQt5.sip', '-t', 'Qt_5_15_0', '-t', 'WS_X11', 'rviz.sip']' returned non-zero exit status 1.
make[2]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:108: sip/rviz_sip/Makefile] Error 1
make[1]: *** [CMakeFiles/Makefile2:4686: src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 56%] Built target rviz_default_plugin_autogen
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
error: target not found: ignition-cmake
error: target not found: ignition-tools
error: target not found: ros-noetic-controller-manager-msgs
 -> error making: ros-noetic-rviz
```

Steps to reproduce:
- Arch rolling updated
- No packages installed from Arch4edu repo - simple AUR installation.

AchmadFathoni, 2021-12-19:
Yes. I got the same error with `ros-noetic-qt-gui-cpp`.

AchmadFathoni, 2021-12-19:
This the result when I run directly into terminal the command that python-subprocess run:
```
sip -c build/sip/qt_gui_cpp_sip -b build/sip/qt_gui_cpp_sip/pyqtscripting.sbf -I /usr/lib/python3.10/site-packages/PyQt5/bindings -w -n PyQt5.sip -t Qt_5_15_0 -t WS_X11 qt_gui_core-0.4.2/qt_gui_cpp/src/qt_gui_cpp_sip/qt_gui_cpp.sip
sip: Deprecation warning: qt_gui_core-0.4.2/qt_gui_cpp/src/qt_gui_cpp_sip/qt_gui_cpp.sip:1: %Module version number should be specified using the 'version' argument
sip: /usr/lib/python3.10/site-packages/PyQt5/bindings/QtCore/QtCoremod.sip:23: syntax error
```

AchmadFathoni, 2021-12-19:
Related to https://github.com/ros-noetic-arch/ros-noetic-rviz/issues/4#issue-1041691468

AchmadFathoni, 2021-12-19:
Arch package maintainer give up sip4 in their pyqt5 package
https://github.com/archlinux/svntogit-packages/commit/9f73b4aabafa235823c529e3a37799ce678b776d#diff-3e341d2d9c67be01819b25b25d5e53ea3cdf3a38d28846cda85a195eb9b7203a
Maybe we should make custom pyqt5 package in AUR that include the sip4 patch

AchmadFathoni, 2021-12-21:
I've made custom pyqt5 package that include deleted sip4 patch. Let me know if I need to fix something
```
# Maintainer : Achmad Fathoni <fathoni.id@gmail.com>    
# Contributor: Antonio Rojas <arojas@archlinux.org>     
# Contributor: Felix Yan <felixonmars@archlinux.org>   
# Contributor: Andrea Scarpino <andrea@archlinux.org>ton
# Contributor: Yichao Yu <yyc1992@gmail.com>
# Contributor: Douglas Soares de Andrade <douglas@archlinux.org>
# Contributor: riai <riai@bigfoot.com> Ben <ben@benmazer.net>

pkgbase=pyqt5
pkgname=('python-pyqt5-sip4-patch')
pkgdesc="python-pyqt5 with sip4 support"
pkgver=5.15.6
pkgrel=1
arch=('x86_64')
url="https://riverbankcomputing.com/software/pyqt/intro"
license=('GPL')
groups=(pyqt5)
depends=('python-pyqt5-sip' 'qt5-base')
optdepends=('python-opengl: enable OpenGL 3D graphics in PyQt applications'
            'python-dbus: for python-dbus mainloop support'
            'qt5-multimedia: QtMultimedia, QtMultimediaWidgets'
            'qt5-tools: QtHelp, QtDesigner'
            'qt5-svg: QtSvg'
            'qt5-webkit: QtWebKit, QtWebKitWidgets'
            'qt5-xmlpatterns: QtXmlPatterns'
            'qt5-declarative: QtQml, qmlplugin'
            'qt5-serialport: QtSerialPort'
            'qt5-websockets: QtWebSockets'
            'qt5-connectivity: QtNfc, QtBluetooth'
            'qt5-x11extras: QtX11Extras'
            'qt5-remoteobjects: QtRemoteObjects'
            'qt5-speech: QtTextToSpeech'
            'qt5-quick3d: QtQuick3D')
provides=('qt5-python-bindings' 'python-pyqt5')
makedepends=('sip' 'pyqt-builder' 'python-opengl' 'python-dbus'
             'qt5-connectivity' 'qt5-multimedia' 'qt5-tools' 'qt5-serialport' 'qt5-speech' 'qt5-svg'
             'qt5-webkit' 'qt5-websockets' 'qt5-x11extras' 'qt5-xmlpatterns' 'qt5-remoteobjects' 'qt5-quick3d')
conflicts=('pyqt5-common' 'python-pyqt5')
source=("https://pypi.python.org/packages/source/P/PyQt5/PyQt5-$pkgver.tar.gz"
         restore-sip4-support.patch)
sha256sums=('80343bcab95ffba619f2ed2467fd828ffeb0a251ad7225be5fc06dcc333af452'
            'c8881a8dc666e509773f93330615d310f9f040c2167dc9e98843741691395f72')

prepare() {
  patch -d PyQt5-$pkgver -p1 < restore-sip4-support.patch # https://www.riverbankcomputing.com/pipermail/pyqt/2021-November/044346.html
}
build() {
  cd PyQt5-$pkgver
  sip-build \
    --confirm-license \
    --no-make \
    --api-dir /usr/share/qt/qsci/api/python
  cd build
  make
}

package(){
  cd PyQt5-$pkgver/build
  make INSTALL_ROOT="$pkgdir" install -j1

  # Remove unused py2 version of uic modules:
  rm -r "$pkgdir"/usr/lib/python*/site-packages/PyQt5/uic/port_v2

  # compile Python bytecode
  python -m compileall -d / "$pkgdir"/usr/lib
  python -O -m compileall -d / "$pkgdir"/usr/lib
}

```
The patch can be downloaded with
`wget https://raw.githubusercontent.com/archlinux/svntogit-packages/e0bceb60af56e406df64dfcb262dfa50f13ddf29/trunk/restore-sip4-support.patch`

I will put this on AUR next week.

NextAlone, 2021-12-22:
> I've made custom pyqt5 package that include deleted sip4 patch. Let me know if I need to fix something
>
> ```
> # Maintainer : Achmad Fathoni <fathoni.id@gmail.com>
> # Contributor: Antonio Rojas [...]

it failed with output belows
```
==> Starting build()...
Traceback (most recent call last):
  File "/usr/bin/sip-build", line 33, in <module>
    sys.exit(load_entry_point('sip==6.5.0', 'console_scripts', 'sip-build')())
  File "/usr/bin/sip-build", line 25, in importlib_load_entry_point
    return next(matches).load()
StopIteration
```

AchmadFathoni, 2021-12-22:
Well you should build it in clean chroot. But if it too complicated you should just download the old official python-pyqt5 and install it
```
$ wget https://archive.archlinux.org/repos/2021/12/13/extra/os/x86_64/python-pyqt5-5.15.6-4-x86_64.pkg.tar.zst
$ sudo pacman -U python-pyqt5-5.15.6-4-x86_64.pkg.tar.zst
```
And that add python-pyqt5 to blacklisted package in `/etc/pacman.conf` to prevent it from being updated.

NextAlone, 2021-12-22:
> Well you should build it in clean chroot. But if it too complicated you should just download the old official python-pyqt5 and install it
>
> ```
> $ wget [...]

i just deleted the diff in the file, and it works too, and thank you too
[image]

closed 2021-12-25 by AchmadFathoni