ros-noetic-arch/ros-noetic-desktop-full #35: Problem in instalarion of package ros-noetic-rviz

focal-lossa, 2022-01-07:
I'm trying to install the ros-noetic-desktop-full AUR package using yay, but the installation process fails in 98%, when Running the SIP generator for rviz_sip Python bindings.

Here is a piece of the output of the command `yay -S ros-noetic-desktop-full`, where the installation process fails:

> [ 95%] Building CXX object src/test/CMakeFiles/render_points_test.dir/__/rviz/ogre_helpers/orbit_camera.cpp.o
> [ 97%] Linking CXX executable ../../devel/lib/rviz/render_points_test
> [ 97%] Built target render_points_test
> [ 97%] Automatic MOC for target two_render_widgets
> [ 97%] Built target two_render_widgets_autogen
> [ 97%] Building CXX object src/test/CMakeFiles/two_render_widgets.dir/two_render_widgets_autogen/mocs_compilation.cpp.o
> [ 97%] Building CXX object src/test/CMakeFiles/two_render_widgets.dir/two_render_widgets.cpp.o
> [ 98%] Linking CXX executable ../../devel/lib/rviz/two_render_widgets
> [ 98%] Built target two_render_widgets
> [ 98%] Running SIP generator for rviz_sip Python bindings...
> sip: /usr/lib/python3.10/site-packages/PyQt5/bindings/QtCore/QtCoremod.sip:23: syntax error
> Traceback (most recent call last):
>   File "/opt/ros/noetic/share/python_qt_binding/cmake/sip_configure.py", line 122, in <module>
>     subprocess.check_call(cmd)
>   File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', '/home/icaro/.cache/yay/ros-noetic-rviz/src/build/sip/rviz_sip', '-b', '/home/icaro/.cache/yay/ros-noetic-rviz/src/build/sip/rviz_sip/pyqtscripting.sbf', '-I', '/usr/lib/python3.10/site-packages/PyQt5/bindings', '-w', '-n', 'PyQt5.sip', '-t', 'Qt_5_15_0', '-t', 'WS_X11', 'rviz.sip']' returned non-zero exit status 1.
> make[2]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:108: sip/rviz_sip/Makefile] Error 1
> make[1]: *** [CMakeFiles/Makefile2:4686: src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2
> make: *** [Makefile:136: all] Error 2
> ==> ERROR: A failure occurred in build().
>     Aborting...
>  -> error making: ros-noetic-rviz

Some informations:
OS: Arch Linux x86_64
Kernel: 5.15.13-arch1-1

I'll provide more information as requested.

AchmadFathoni, 2022-01-07:
Downgrade your python-pyqt5 to 5.15.6-4 or follow https://github.com/ros-noetic-arch/ros-noetic-rviz/issues/4#issuecomment-1000416152

focal-lossa, 2022-01-09:
I used the method of this commit [https://github.com/ros-noetic-arch/ros-noetic-rviz/issues/4#issuecomment-1000416152](url), recommended by you and after installing more dependencies, the installation of ROS was completed.

Thank you so much for your help!

closed 2022-01-09 by AchmadFathoni