ros-noetic-arch/ros-noetic-desktop-full #36: Unflag already up-to-date package

AchmadFathoni, 2022-03-20:
Hi @acxz, please unflag https://aur.archlinux.org/packages/ros-noetic-robot-localization and https://aur.archlinux.org/packages/ros-noetic-geographic-msgs. Someone flagged them because they think the newest version is for ROS1. I just learned that co-maintainer cannot unflag pakage.

acxz, 2022-03-20:
Yep was taking care of them right now. Do you mind sending @carlosal1015 an email (and CC me) to stop doing this? This is not the first time he has done it nor only for ROS packages. He just goes around and flags packages even if he doesn't use them. It is very disrespectful behavior.

AchmadFathoni, 2022-03-20:
I've emailed him some explanation.

closed 2022-03-20 by AchmadFathoni