ros-noetic-arch/ros-noetic-desktop-full #38: build error in sdformat-9

JohannesSchumacher, 2022-04-05:
Hey guys,
when i try to install ros-noetic-desktop-full with yay, it fails when building sdformat-9 with error on line 84 conversionMap’ was not declared in this scope:

`/home/jogi/.cache/yay/sdformat-9/src/sdformat-sdformat9_9.5.0/src/Converter.cc:84:16: Fehler: »conversionMap« wurde in diesem Gültigkeitsbereich nicht definiert
   84 |     fromIter = conversionMap.find(origVersion);
      |                ^~~~~~~~~~~~~
make[2]: *** [src/CMakeFiles/sdformat9.dir/build.make:188: src/CMakeFiles/sdformat9.dir/Converter.cc.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:3233: src/CMakeFiles/sdformat9.dir/all] Fehler 2
make: *** [Makefile:146: all] Fehler 2
==> FEHLER: Ein Fehler geschah in build().
    Breche ab...
 -> Fehler beim Ausführen: sdformat-9
`
Unfortunately i couldn't find any information about this. Do you have any idea for a fix? It's a dependency on gazebo. So I dont know if you can help.. if not im sorry

acxz, 2022-04-05:
the aur package for `sdformat-9` is maintained by [Homalozoa](https://aur.archlinux.org/account/Homalozoa) @Homalozoa

You'll have to post the report to the following package: https://aur.archlinux.org/packages/sdformat-9

closed 2022-04-05 by acxz

AchmadFathoni, 2022-04-05:
@JohannesSchumacher I can't reproduce this problem. Try to build it in [clean-chroot](https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot#Convenience_way)