ros-noetic-arch/ros-noetic-desktop-full #39: Build failed. No rule to make target '/usr/lib/libboost_date_time.so.1.78.0'.

0xNOY, 2022-06-22:
Build failed when installing ros-noetic-desktop using AUR helper yay. I searched about it for several hours but could not reach an answer. I would be glad if you could help me.

```
$ yay -S ros-noetic-desktop
...
make[2]: *** No rule to make target '/usr/lib/libboost_date_time.so.1.78.0', needed by 'devel/lib/librosconsole.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:327: CMakeFiles/rosconsole.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ros-noetic-rosconsole
```
<details>

```
==> Making package: ros-noetic-rosconsole 1.14.3-8 (Wed 22 Jun 2022 06:06:14 PM JST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found ros-noetic-rosconsole-1.14.3.tar.gz
==> Validating source files with sha256sums...
    ros-noetic-rosconsole-1.14.3.tar.gz ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting ros-noetic-rosconsole-1.14.3.tar.gz with bsdtar
==> Sources are ready.
==> Making package: ros-noetic-rosconsole 1.14.3-8 (Wed 22 Jun 2022 06:06:19 PM JST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Removing existing $pkgdir/ directory...
==> Starting build()...
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/noy/.cache/yay/ros-noetic-rosconsole/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python (found suitable version "3.10.5", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.10/site-packages/em.py
-- Using empy: /usr/lib/python3.10/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /home/noy/.cache/yay/ros-noetic-rosconsole/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found Python: /usr/bin/python3.10 (found version "3.10.5") found components: Interpreter
-- Found Threads: TRUE
-- Using Python nosetests: /usr/bin/nosetests-3.10
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Found Boost: /usr/lib64/cmake/Boost-1.79.0/BoostConfig.cmake (found version "1.79.0") found components: regex system thread
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.0")
-- rosconsole backend: glog
-- Configuring done
-- Generating done
-- Build files have been written to: /home/noy/.cache/yay/ros-noetic-rosconsole/src/build
[ 16%] Building CXX object CMakeFiles/rosconsole_backend_interface.dir/src/rosconsole/rosconsole_backend.cpp.o
[ 33%] Linking CXX shared library devel/lib/librosconsole_backend_interface.so
[ 33%] Built target rosconsole_backend_interface
[ 50%] Building CXX object CMakeFiles/rosconsole_glog.dir/src/rosconsole/impl/rosconsole_glog.cpp.o
[ 66%] Linking CXX shared library devel/lib/librosconsole_glog.so
[ 66%] Built target rosconsole_glog
[ 83%] Building CXX object CMakeFiles/rosconsole.dir/src/rosconsole/rosconsole.cpp.o
make[2]: *** No rule to make target '/usr/lib/libboost_date_time.so.1.78.0', needed by 'devel/lib/librosconsole.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:327: CMakeFiles/rosconsole.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ros-noetic-rosconsole
```

</details>

AchmadFathoni, 2022-06-22:
Try this, it will rebuild all ROS package that depends on boost
`yay -S $(pactree boost -r -l | tail +2 | grep 'ros-noetic') --sudoloop --noconfirm --rebuildall`

AchmadFathoni, 2022-06-22:
Revision: I think you should rebuild non ROS package first with
`yay -S $(pactree boost -r -l -u | grep 'ros-noetic' -v) --sudoloop --noconfirm --rebuildall`
Then rebuild ROS package with
`yay -S $(pactree boost -r -l -u | grep 'ros-noetic') --sudoloop --noconfirm --rebuildall`

AchmadFathoni, 2022-06-24:
> Revision: I think you should rebuild non ROS package first with `yay -S $(pactree boost -r -l -u | grep 'ros-noetic' -v) --sudoloop --noconfirm --rebuildall` Then rebuild ROS package with `yay -S $(pactree boost -r -l -u | grep 'ros-noetic') --sudoloop --noconfirm --rebuildall`

Revision#2:
should change `boost` to `boost-libs`

acxz, 2022-07-13:
@0xNOY please comment if you are still facing this issue. I'll close it for now.

closed 2022-07-13 by acxz