ros-noetic-arch/ros-noetic-desktop-full #40: install will make my laptop screen black

BackMountainDevil, 2022-09-21:
I try to install it via pamac(ros-noetic-desktop-full pkgver='1.5.0') on Manjaro twice. Everytime my pc screen turn black, ubable to use keyboard and mouse to shutdown pc. I can only press the power button to force to shutdown.

according to pamac history,  the last pkg is ros-noetic-pcl-msgs

BackMountainDevil, 2022-09-21:
AMD R5 4600u + 16GB Ram

<details>
<summary>System Info</summary>

```bash
$ uname -a
Linux 82dm 5.15.65-1-MANJARO #1 SMP PREEMPT Mon Sep 5 10:15:47 UTC 2022 x86_64 GNU/Linux

$ inxi --admin --verbosity=7 --filter --width
System:
  Kernel: 5.15.65-1-MANJARO arch: x86_64 bits: 64 compiler: gcc v: 12.2.0
    parameters: BOOT_IMAGE=/boot/vmlinuz-5.15-x86_64
    root=UUID=5d6e2a51-4876-44ba-9f6e-acc3e0bf2767 rw quiet
    udev.log_priority=3
  Desktop: KDE Plasma v: 5.25.5 tk: Qt v: 5.15.5 wm: kwin_x11 vt: 1 dm: SDDM
    Distro: Manjaro Linux base: Arch Linux
Machine:
  Type: Laptop System: LENOVO product: 82DM v: Lenovo XiaoXinPro-13ARE 2020
     Chassis: type: 10 v: Lenovo XiaoXinPro-13ARE
    2020 
  Mobo: LENOVO model: LNVNB161216 v: SDK0L77769WIN
     UEFI: LENOVO v: F0CN36WW date: 05/16/2022
Battery:
  ID-1: BAT0 charge: 49.0 Wh (100.0%) 
    volts: 12.9 min: 11.5 model: SMP L19M3PD3 type: Li-poly 
    status: full cycles: 247
Memory:
  RAM: total: 14.99 GiB used: 3.67 GiB (24.5%)
  RAM Report: permissions: Unable to run dmidecode. Root privileges
    required.
CPU:
  Info: model: AMD Ryzen 5 4600U with Radeon Graphics bits: 64 type: MT MCP
    arch: Zen 2 gen: 3 level: v3 built: 2020-22 process: TSMC n7 (7nm)
    family: 0x17 (23) model-id: 0x60 (96) stepping: 1 microcode: 0x8600106
  Topology: cpus: 1x cores: 6 tpc: 2 threads: 12 smt: enabled cache:
    L1: 384 KiB desc: d-6x32 KiB; i-6x32 KiB L2: 3 MiB desc: 6x512 KiB L3: 8 MiB
    desc: 2x4 MiB
  Speed (MHz): avg: 1507 high: 2292 min/max: 1400/2100 boost: disabled
    scaling: driver: acpi-cpufreq governor: schedutil cores: 1: 1397 2: 1397
    3: 2292 4: 1587 5: 1565 6: 1397 7: 1397 8: 1474 9: 1397 10: 1397 11: 1398
    12: 1392 bogomips: 50320
  Flags: 3dnowprefetch abm adx aes aperfmperf apic arat avic avx avx2 bmi1
    bmi2 bpext cat_l3 cdp_l3 clflush clflushopt clwb clzero cmov cmp_legacy
    constant_tsc cpb cpuid cqm cqm_llc cqm_mbm_local cqm_mbm_total
    cqm_occup_llc cr8_legacy cx16 cx8 de decodeassists extapic extd_apicid
    f16c flushbyasid fma fpu fsgsbase fxsr fxsr_opt ht hw_pstate ibpb ibrs ibs
    irperf lahf_lm lbrv lm mba mca mce misalignsse mmx mmxext monitor movbe
    msr mtrr mwaitx nonstop_tsc nopl npt nrip_save nx osvw overflow_recov pae
    pat pausefilter pclmulqdq pdpe1gb perfctr_core perfctr_llc perfctr_nb
    pfthreshold pge pni popcnt pse pse36 rapl rdpid rdpru rdrand rdseed rdt_a
    rdtscp rep_good sep sha_ni skinit smap smca smep ssbd sse sse2 sse4_1
    sse4_2 sse4a ssse3 stibp succor svm svm_lock syscall tce topoext tsc
    tsc_scale umip v_spec_ctrl v_vmsave_vmload vgif vmcb_clean vme vmmcall
    wbnoinvd wdt xgetbv1 xsave xsavec xsaveerptr xsaveopt xsaves
  Vulnerabilities:
  Type: itlb_multihit status: Not affected
  Type: l1tf status: Not affected
  Type: mds status: Not affected
  Type: meltdown status: Not affected
  Type: mmio_stale_data status: Not affected
  Type: retbleed mitigation: untrained return thunk; SMT enabled with STIBP
    protection
  Type: spec_store_bypass mitigation: Speculative Store Bypass disabled via
    prctl and seccomp
  Type: spectre_v1 mitigation: usercopy/swapgs barriers and __user pointer
    sanitization
  Type: spectre_v2 mitigation: Retpolines, IBPB: conditional, STIBP:
    always-on, RSB filling, PBRSB-eIBRS: Not affected
  Type: srbds status: Not affected
[... 78 lines cut ...]
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 4 rev: 2.0
    speed: 480 Mb/s chip-ID: 1d6b:0002 class-ID: 0900
  Device-1: 1-4:2 info: Realtek Bluetooth Radio type: Bluetooth
    driver: btusb interfaces: 2 rev: 1.0 speed: 12 Mb/s power: 500mA
    chip-ID: 0bda:c123 class-ID: e001 
  Hub-2: 2-0:1 info: Super-speed hub ports: 2 rev: 3.1 speed: 10 Gb/s
    chip-ID: 1d6b:0003 class-ID: 0900
  Hub-3: 3-0:1 info: Hi-speed hub with single TT ports: 4 rev: 2.0
    speed: 480 Mb/s chip-ID: 1d6b:0002 class-ID: 0900
  Device-1: 3-2:2 info: Chicony Integrated Camera type: Video
    driver: uvcvideo interfaces: 4 rev: 2.0 speed: 480 Mb/s power: 500mA
    chip-ID: 04f2:b67c class-ID: 0e02 
  Hub-4: 4-0:1 info: Super-speed hub ports: 2 rev: 3.1 speed: 10 Gb/s
    chip-ID: 1d6b:0003 class-ID: 0900
Sensors:
  System Temperatures: cpu: 49.2 C mobo: N/A gpu: amdgpu temp: 43.0 C
  Fan Speeds (RPM): N/A
Info:
  Processes: 302 Uptime: 2h 48m wakeups: 1 Init: systemd v: 251
  default: graphical tool: systemctl Compilers: gcc: 12.2.0 clang: 14.0.6
  Packages: pm: pacman pkgs: 1235 libs: 318 tools: pamac pm: flatpak pkgs: 0
  Shell: Bash v: 5.1.16 running-in: codium inxi: 3.3.21
```
</details>

acxz, 2022-09-21:
We don't officially support Manjaro, since it is not AUR compatible.

Can you try a Arch binary compatible distro instead? Most of our previous Manjaro users switch over to [EndeavourOS](https://endeavouros.com/) to get the ros packages working from the AUR. Let us know if that resolves the issue for you.

BackMountainDevil, 2022-09-21:
Oh, god, Manjaro is not AUR compatible? I will try that tonight

BackMountainDevil, 2022-09-22:
I have switched over EndevaourOS. And I try `yay ros-noetic-desktop-full -Y`. From last night to this morning. something wrong happen

```bash
-- Could not find the VTK package due to a missing dependency: PDAL
COMPONENT = openvdb
-- Found OpenVDB: /usr/include (found version "9.1.0") found components: openvdb 
-- OpenVDB ABI Version: 9
CMake Error at /usr/lib64/cmake/Boost-1.80.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /usr/lib64/cmake/boost_iostreams-1.80.0/boost_iostreams-config.cmake

  but it set boost_iostreams_FOUND to FALSE so package "boost_iostreams" is
  considered to be NOT FOUND.  Reason given by package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_iostreams.so.1.80.0 (release runtime,
  Boost_USE_DEBUG_RUNTIME=TRUE)

  * libboost_iostreams.a (static, Boost_USE_STATIC_LIBS=OFF)

Call Stack (most recent call first):
  /usr/lib64/cmake/Boost-1.80.0/BoostConfig.cmake:262 (boost_find_component)
  /usr/share/cmake/Modules/FindBoost.cmake:594 (find_package)
  /usr/share/cmake/Modules/FindOpenVDB.cmake:499 (find_package)
  /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:650 (find_package)
  /usr/lib64/cmake/vtk/vtk-config.cmake:150 (include)
  cmake/pcl_find_vtk.cmake:25 (find_package)
  CMakeLists.txt:375 (include)

-- Configuring incomplete, errors occurred!
See also "/home/kearney/.cache/yay/pcl/src/build/CMakeFiles/CMakeOutput.log".
==> 错误： 在 build() 中发生一个错误。
    正在放弃...
 -> 生成时出错: pcl

```

BackMountainDevil, 2022-09-22:
I try yay ros-noetic-desktop-full -Y again after set Lang to English and got these:

```
CMake Warning at /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:209 (find_package):
  By not providing "FindCLI11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "CLI11", but
  CMake did not find one.

  Could not find a package configuration file provided by "CLI11" with any of
  the following names:

    CLI11Config.cmake
    cli11-config.cmake

  Add the installation prefix of "CLI11" to CMAKE_PREFIX_PATH or set
  "CLI11_DIR" to a directory containing one of the above files.  If "CLI11"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:150 (include)
  cmake/pcl_find_vtk.cmake:25 (find_package)
  CMakeLists.txt:375 (include)

-- Could not find the VTK package due to a missing dependency: CLI11
-- Found Embree v3.13.5: /usr/lib/libembree3.so.3.13.5
-- Found Open VKL v1.3.0: /usr/lib/libopenvkl.so.1.3.0
-- Found OSPRay: /usr
-- Found OpenVR: /usr/lib/libopenvr_api.so  
-- Found FontConfig: /usr/lib/libfontconfig.so  
-- Found Python: /usr/bin/python3.10 (found version "3.10.7") found components: Interpreter 
-- Found mpi4py: /usr/lib/python3.10/site-packages/mpi4py/include  
-- Could NOT find PostgreSQL (missing: PostgreSQL_LIBRARY PostgreSQL_INCLUDE_DIR) 
-- Could not find the VTK package due to a missing dependency: PostgreSQL
-- Could NOT find Boost: Found unsuitable version "1.80.0", but required is exact version "1.79.0" (found /usr/include, )
-- Could not find the VTK package due to a missing dependency: Boost
CMake Warning at /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:603 (find_package):
  By not providing "FindPDAL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PDAL", but
  CMake did not find one.

  Could not find a package configuration file provided by "PDAL" with any of
  the following names:

    PDALConfig.cmake
    pdal-config.cmake

  Add the installation prefix of "PDAL" to CMAKE_PREFIX_PATH or set
  "PDAL_DIR" to a directory containing one of the above files.  If "PDAL"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:150 (include)
  cmake/pcl_find_vtk.cmake:25 (find_package)
  CMakeLists.txt:375 (include)

-- Could not find the VTK package due to a missing dependency: PDAL
COMPONENT = openvdb
-- Found OpenVDB: /usr/include (found version "9.1.0") found components: openvdb 
-- OpenVDB ABI Version: 9
CMake Error at /usr/lib64/cmake/Boost-1.80.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /usr/lib64/cmake/boost_iostreams-1.80.0/boost_iostreams-config.cmake

  but it set boost_iostreams_FOUND to FALSE so package "boost_iostreams" is
  considered to be NOT FOUND.  Reason given by package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_iostreams.so.1.80.0 (release runtime,
  Boost_USE_DEBUG_RUNTIME=TRUE)

  * libboost_iostreams.a (static, Boost_USE_STATIC_LIBS=OFF)

Call Stack (most recent call first):
  /usr/lib64/cmake/Boost-1.80.0/BoostConfig.cmake:262 (boost_find_component)
  /usr/share/cmake/Modules/FindBoost.cmake:594 (find_package)
  /usr/share/cmake/Modules/FindOpenVDB.cmake:499 (find_package)
  /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:650 (find_package)
  /usr/lib64/cmake/vtk/vtk-config.cmake:150 (include)
  cmake/pcl_find_vtk.cmake:25 (find_package)
  CMakeLists.txt:375 (include)

-- Configuring incomplete, errors occurred!
See also "/home/kearney/.cache/yay/pcl/src/build/CMakeFiles/CMakeOutput.log".
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: pcl
[kearney@82dm ~]$ 

```

acxz, 2022-09-22:
ok, at least we have an error message that we can debug now.

>  `-> 生成时出错: pcl`

This is a problem with the `pcl` package. See the comments here: https://aur.archlinux.org/packages/pcl#comment-880849

acxz, 2022-09-22:
I'm gonna close this issue as a duplicate of #41 so that we can consolidate the issues.

closed 2022-09-22 by acxz