ros-noetic-arch/ros-noetic-desktop-full #43: Not able  to install with newest poco 1.12.4-1

TangdizhiH, 2022-12-18:
Hi, dear maintainer of the ros-noetic-arch. As suggested in the title. Some of the packages in `ros-noetic-desktop-full` relies on the `/usr/lib/libPocoFoundation.so.91`. Which is removed with the newest poco version in arch repository.

Reproduce: try to install `ros-noetic-desktop-full` on archlinux.

Expected behavior: successfully installed

Actual behavior: the installation fails due to `libPocoFoundation.so.91` not found.

Also, this problem also happens when I try to install `ros-noetic-turtlebot3` meta package.

Thank you for your effort to make ros noetic available for arch users.

AchmadFathoni, 2023-02-19:
You can try install ros-noetic via conda environment [here](https://robostack.github.io/GettingStarted.html)

acxz, 2023-03-07:
If you still want to stick to the ros-noetic-arch packages, the current suggested solution is the following: https://github.com/ros-noetic-arch/ros-noetic-desktop-full/issues/10#issuecomment-653168991

closed 2023-03-07 by acxz