ros-noetic-arch/ros-noetic-desktop-full #51: Problema to install ros-noetic-desktop-full

davelm31, 2024-12-14:
Hi, I have some problem when I try to install ros noetic `yay -Syu ros-noetic-desktop-full --rebuildtree`. 
The error is when trying to install the package `ros-noetci-catkin` and that generates a chain of errors
`==> Making package: ros-noetic-catkin 0.8.10-3 (Sat 14 Dec 2024 04:18:29 AM UTC)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found ros-noetic-catkin-0.8.10.tar.gz
==> Validating source files with sha256sums...
    ros-noetic-catkin-0.8.10.tar.gz ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting ros-noetic-catkin-0.8.10.tar.gz with bsdtar
==> Sources are ready.
==> Making package: ros-noetic-catkin 0.8.10-3 (Sat 14 Dec 2024 04:18:32 AM UTC)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Removing existing $pkgdir/ directory...
==> Starting build()...
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- The C compiler identification is GNU 14.2.1
-- The CXX compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/dace/.cache/yay/ros-noetic-catkin/src/build/devel
-- Using CMAKE_PREFIX_PATH:
CMake Warning (dev) at cmake/python.cmake:4 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  cmake/all.cmake:164 (include)
  CMakeLists.txt:8 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python (found version "3.12.7")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.12/site-packages/em.py
-- Using empy: /usr/lib/python3.12/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dace/.cache/yay/ros-noetic-catkin/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
CMake Warning at /usr/src/gmock/CMakeLists.txt:40 (project):
  VERSION keyword not followed by a value or was followed by a value that
  expanded to nothing.

CMake Warning at /usr/src/googletest/CMakeLists.txt:50 (project):
  VERSION keyword not followed by a value or was followed by a value that
  expanded to nothing.

-- Found Threads: TRUE
CMake Error at /usr/src/googletest/CMakeLists.txt:124 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at /usr/src/googletest/CMakeLists.txt:142 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at /usr/src/gmock/CMakeLists.txt:99 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at /usr/src/gmock/CMakeLists.txt:102 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

-- Using Python nosetests: /usr/bin/nosetests-3.12
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Configuring incomplete, errors occurred!
==> ERROR: A failure occurred in build().
    Aborting...
 -> Failed to install layer, rolling up to next layer.error:error making: ros-noetic-catkin - exit status 4
==> Making package: ros-noetic-catkin 0.8.10-3 (Sat 14 Dec 2024 04:18:41 AM UTC)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found ros-noetic-catkin-0.8.10.tar.gz
==> Validating source files with sha256sums...
    ros-noetic-catkin-0.8.10.tar.gz ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting ros-noetic-catkin-0.8.10.tar.gz with bsdtar
==> Sources are ready.
==> Making package: ros-noetic-catkin 0.8.10-3 (Sat 14 Dec 2024 04:18:43 AM UTC)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Removing existing $pkgdir/ directory...
==> Starting build()...
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- The C compiler identification is GNU 14.2.1
-- The CXX compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/dace/.cache/yay/ros-noetic-catkin/src/build/devel
-- Using CMAKE_PREFIX_PATH:
CMake Warning (dev) at cmake/python.cmake:4 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  cmake/all.cmake:164 (include)
  CMakeLists.txt:8 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python (found version "3.12.7")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/lib/python3.12/site-packages/em.py
-- Using empy: /usr/lib/python3.12/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dace/.cache/yay/ros-noetic-catkin/src/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
CMake Warning at /usr/src/gmock/CMakeLists.txt:40 (project):
  VERSION keyword not followed by a value or was followed by a value that
  expanded to nothing.

CMake Warning at /usr/src/googletest/CMakeLists.txt:50 (project):
  VERSION keyword not followed by a value or was followed by a value that
  expanded to nothing.

-- Found Threads: TRUE
CMake Error at /usr/src/googletest/CMakeLists.txt:124 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at /usr/src/googletest/CMakeLists.txt:142 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at /usr/src/gmock/CMakeLists.txt:99 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at /usr/src/gmock/CMakeLists.txt:102 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

-- Using Python nosetests: /usr/bin/nosetests-3.12
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Configuring incomplete, errors occurred!
==> ERROR: A failure occurred in build().
    Aborting...
 -> Failed to install layer, rolling up to next layer.error:error making: ros-noetic-catkin - exit status 4
 `

jnvandermeer, 2025-01-27:
See the AUR page, installing with yay/AUR the googletest-git seems to make some progress (and it deals with this issue)

jnvandermeer, 2025-01-29:
... but there are a phletora of other issues

davelm31, 2025-04-04:
This does not work

jnvandermeer, 2025-04-04:
Yes, in the end I sort of gave up and installed Ubuntu 20.04 and followed those installation instructions. I think currently with arch there is a bit of too much dependency issues. This worked for me. It's not a solution to this of course :)